pimoroni/explorer-hat #2: Input.on_changed doesn't fir correctly

mr-eggleton, 2015-03-19:
def handle_callback(pin):
 if self.read() == 1 and callable(self.handle_pressed):
  self.handle_pressed(self)
 elif self.read() == 0 and callable(self.handle_released):
  self.handle_released(self)
 if callable(self.handle_changed):  # SURELY this should check has_changed or self.last
  self.handle_changed(self)

also why are we calling read that often?

Gadgetoid, 2015-03-25:
The actual edge detection happens way down in RPi.GPIO, so I think this may be a de-bouncing problem.

The function `handle_callback` is executed every time RPi.GPIO handles an edge transition- whether it's rising or falling.

`self.read() == 1` checks if the transition was a rising edge, and calls `handle_pressed` if it is. This happens once with each transition that RPi.GPIO detects. Similarly reading the pin state at 0 detects a falling edge.

There are many reasons why this is slightly less than ideal code, but you can't attach both a rising and falling edge event detect to RPi.GPIO without using GPIO.BOTH:

```python
>>> gpio.add_event_detect(23, gpio.RISING, callback=high)
>>> gpio.add_event_detect(23, gpio.FALLING, callback=low)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
RuntimeError: Conflicting edge detection already enabled for this GPIO channel
```

You also can't supply RPi.GPIO with a function that accepts the pin state as a separate parameter when using a handler for `BOTH`.

So the only choice left is to read the pin state immediately afterwards- which could potentially miss a transition if it's shorter than the time it takes RPi.GPIO to catch an edge, call the Python handler it's been passed and for that Python handler to read the pin state again.

Can you put a scope on your rotary encoder? It would be interesting to know exactly what the signal looks like and how frequent the high/low transitions are- it's certainly not impossible that they are too short for the library to see.

Out of interest, try using RPi.GPIO's add_event_detect directly and see what you get:

```python
import RPi.GPIO as GPIO
INPUT_ONE = 23
GPIO.setmode(GPIO.BCM)
GPIO.setup(INPUT_ONE, GPIO.IN)
def pin_changed(pin):
    print(pin, GPIO.input(pin) == 1)
GPIO.add_event_detect(INPUT_ONE, GPIO.BOTH, callback=pin_changed, bouncetime=1)
```

mr-eggleton, 2015-03-27:
In my case the event is constantly triggering when something is in front of my reflection sensor and doing nothing the rest of the time. I will shorten the bouncetime to see if its fluttering but I suspect my electronics are wrong.

closed 2015-03-27 by mr-eggleton