pimoroni/explorer-hat #3: Python 3?

MissPhilbin, 2015-03-23:
Instructions are for Python 2 and do not include Python 3. Would be useful for all users if both were included. Many thanks.

Gadgetoid, 2015-03-23:
To my knowledge there's still no Python 3 compatible smbus module in Raspbian, so Explorer HAT wont work with Python 3.

On the Basic the Inputs/Outputs would work, but the cap touch wouldn't, so it would just be confusing to have Python 3 instructions that result in a partially working board.

When we finally get a Python 3 smbus, we'll have to run back through our lineup, make sure it actually does work, and update the software/instructions as necessary.

MissPhilbin, 2015-03-23:
It was my understanding that python3 smbus was being dealt with. My bad. I've pinged members of the PSF to help.

Gadgetoid, 2015-03-23:
Thank you! It was my understanding also, but it's been a long time in the works!

bennuttall, 2015-09-20:
This can be closed, right?

Gadgetoid, 2015-10-15:
Yup!

closed 2015-10-15 by Gadgetoid