pimoroni/explorer-hat #15: Node support?

johnwargo, 2016-02-17:
How do I interact with the Explorer Hat using Node?  Can I just write to the outputs for example using an existing node-based GPIO library or do I have to do something special?

Gadgetoid, 2016-02-23:
You could do some things with a standard NodeJS GPIO library, these include:

* Read digital inputs 1, 2, 3, 4
* Set digital outputs 1, 2, 3, 4
* Toggle LEDs
* Drive the motor channels

You can find a reference of the GPIO pins used for each of these here: https://github.com/pimoroni/explorer-hat/blob/master/documentation/GPIO-pins.md

To use the Analogue channels, or the Capacitive Touch inputs you will need to find, or write, a NodeJS driver for the ADS1015 ( Analog ) or the CAP11xx ( Cap Touch ).

A NodeJS Cap11xx driver already exists for the Display-o-Tron HAT, which you could re-purpose: https://github.com/jorisvervuurt/JVSDisplayOTron

punitganshani, 2016-04-09:
It would be good if you could provide node libraries as well

closed 2016-07-06 by Gadgetoid