pimoroni/explorer-hat #27: Installer fails

JamesBradley86, 2016-10-19:
curl -sS get.pimoroni.com/explorerhat | bash

Fails with the following errors:

W: There is no public key available for the following key IDs:
1397BC53640DB551
W: Failed to fetch http://dl.google.com/linux/chrome/deb/dists/stable/Release  Unable to find expected entry 'main/binary-armhf/Packages' in Release file (Wrong sources.list entry or malformed file)

Any ideas? I suspect it's bloomin' google deleting/moving things around again?

JamesBradley86, 2016-10-19:
(sudo) curl get.pimoroni.com/i2c | bash also fails with the error:

Install must be run as root. Try 'sudo ./i2c'

RogueM, 2016-10-19:
You need to tell us what OS you are running, it's not standard Raspbian behaviour, so I suspect you are running something else?

RogueM, 2016-10-19:
but, yes, the i2c script is mot meant to be called directly, so elevated privileges is required if doing so:

`curl https://get.pimoroni.com/ i2c | sudo bash`

JamesBradley86, 2016-10-19:
Ah. I am running Raspian actually... I've enabled i2c manually, then the rest of the manual install process went fine. I don't *actually* have an explorer hat yet, it's coming tomorrow, so I will report back then as to whether all is well or not, but yes, I am on Raspian... What's going on with the auto-install error?

RogueM, 2016-10-19:
Well, it's not recognising a gpg key, and while I haven't checked I doubt this is the Raspbian one. You should inspect your apt sources file and comment out everything not to do with official repo, if only temporarily:

/etc/apt/sources.list

JamesBradley86, 2016-10-19:
All I've got in there is:

deb http://mirrordirector.raspbian.org/raspbian/ wheezy main contrib non-free rpi

?

RogueM, 2016-10-19:
I'm not sure, maybe the Wheezy key has been revoked, I doubt it, but without checking what ID that key is for that's a wild (if unlikely correct) guess.

Incidentally, while we don't expect Wheezy will be a problem, it's certainly rarely asserted while making tweaks to the scripts... any reason why you need to stick to Wheezy?

JamesBradley86, 2016-10-19:
My SD card reader is knackered, and possibly just because I'm a bit of a Linux noob, I assumed it wasn't possible to upgrade directly from the Pi?

RogueM, 2016-10-19:
You should be able to, though it is a longwinded process, and not without risk.

According to my search your error has to do with google chromium, which spams some secondary apt source file, and tries to fetch a public key that no longer auto-validates.

RogueM, 2016-10-19:
check, /etc/apt/sources.list.d/
... there's probably a file in there with a google related entry. Alternatively I think it would be possible to acknowledge the key if you can get your hands on it (the Ubuntu forums have links that may or may not lead to the public key and allow you to download it).

RogueM, 2016-10-19:
And here's a link to upgrade Wheezy to Jessie:

https://www.raspberrypi.org/forums/viewtopic.php?f=66&t=121880

but you'll have to resolve the google key issue first, or you won't be able to update your indexes... this is what is throwing the error in our installer incidentally, and it's really a basic day-to-day maintenance task, so even if you don't want to upgrade to jessie using this - long and touchy - approach you should try to resolve it, you'll run into similar trouble regularly if not.

my advise though is to buy a current NOOBS card, or replace your SD card reader though - that will save you hours and is a much safer upgrade path.

JamesBradley86, 2016-10-19:
Deleting that file seemed to do the trick... Though I did get this error:

pip-3.2: error: No command by the name pip-3.2 list
  (maybe you meant "pip-3.2 install list")
Unable to install smbus for python 3!

Is this going to be a problem, if I'm happy enough sticking to python 2?

RogueM, 2016-10-19:
It's likely you haven't upgraded your software for a very long while, My guess would be you have python 3.1 which I believe initially shipped with Wheezy.

Run:
`curl https://get.pimoroni.com/ uptodate | bash`

This will update software using the Wheezy sources. A new kernel will be rolled in but it is a fairly safe process. It shouldn't take more than an hour and offset most issues you'd potentially run into trying to upgrade to jessie (a clean SD is a much better option, if you want to consider it at a later date).

JamesBradley86, 2016-10-19:
I'm on python 2.7 (I think!) - thank you, I'll try the above script!

RogueM, 2016-10-19:
Yes, python 2.7 is the latest in that branch, Wheezy was shipping with both 2.7 and 3.1 - then later 3.2.

what do you get if you type:

`python3 --version`

JamesBradley86, 2016-10-20:
I bit the bullet, bought a new SD card reader and installed Jessie :) I don't suppose you could help me to hook this up: http://cdn.sparkfun.com/datasheets/Wireless/General/TWS-BS-3_433.92MHz_ASK_RF_Transmitter_Module_Data_Sheet.pdf to the explorer hat? I'm mostly unsure of what to connect the data in pin to? With an LED, I connect the LED in to 5v and LED out to output 1 (which switches between ground and no ground) but I'm not having any luck getting this to work? The chip outputs a constant 1/on when data in is connected to 5v but I'm wanting to control this from python? Thank you in advance!

closed 2017-04-12 by Gadgetoid