pimoroni/explorer-hat #30: Motor.one()

poilverf, 2016-11-17:
Hello
I arrive to command the two motors
explorerhat.motor.forwards()

But if I try to command each motor with
explorerhat.motor.one.forwards()
it doesn't work
Have you got an idea of the problem ?

RogueM, 2016-11-17:
Is this with the pHAT or the HAT? what is the output, anything? or does it take the command but the motor does not get into motion?

any details about your setup and observation you can share is helpful for understanding what may be happening.

poilverf, 2016-11-17:
I work with the explorer hat pro on a PI3 with jessie, I try wth Python 3.4.2 and Python 2.7.9

If I try :
import time
import explorerhat
print("""
Press CTRL+C to exit.
""")
explorerhat.motor.forwards()
time.sleep(1)
explorerhat.motor.backwards()
time.sleep(1)
explorerhat.motor.forwards(0)
time.sleep(1)

The two motors move and I have the message 
Explorer HAT Pro detected...
Press CTRL+C to exit.

but with

import time
import explorerhat
print("""
Press CTRL+C to exit.
""")
explorerhat.motor.one.forwards()
time.sleep(1)
explorerhat.motor.one.backwards()
time.sleep(1)
explorerhat.motor.one.forwards(0)
time.sleep(1)

I have the message 
Explorer HAT Pro detected...
Press CTRL+C to exit.
I have no error but the motor 1 don't move
It's the same thing with motor 2

Thanks for your help

RogueM, 2016-11-17:
Hum, that is a very odd issue you have there... I doubt that is a factor but are your motors the same polarity? anything peculiar about how you hooked them up to the HAT?

AndiCui, 2016-11-17:
Sounds like wrong wiring to me.
Are both wire of the motor goes into the slots meant for the SAME motor?
Pls double check

poilverf, 2016-11-17:
I double check tomorrow !

poilverf, 2016-11-18:
Hi
Congratulations. You are right.
I am sorry to have made such a big mistake but I am a newbie in raspberry and python so I thought it came from my code
Thanks for your help

closed 2016-11-18 by poilverf