pimoroni/explorer-hat #31: Not Detecting Explorer Hat

riveranl, 2017-04-13:
I believe that I have installed all of the necessary files, however, when I go to python and `import explorerhat` it does nothing. Am I missing a file or something? I even said in the Pimoroni dashboard that I am using an explorerhat pro.

riveranl, 2017-04-14:
In terminal I have run `curl https://get.pimoroni.com/explorerhat | bash`, `sudo apt-get install pimoroni` and, `sudo apt-get install python3-explorerhat`. Then I go to python and run `import explorerhat` in the shell and nothing happens. Shouldn't it say "Explorer HAT Pro detected!"?

Gadgetoid, 2017-04-18:
Have you tried to toggle any LEDs, or use any features of Explorer HAT?

The following code outlines the various detection methods you might see, it has an `else` condition which throws an error and exits:

https://github.com/pimoroni/explorer-hat/blob/master/library/explorerhat/__init__.py#L714-L726

If you're using IDLE, its possible that this condition is being hit, but you're not seeing the message.

You should double-check i2c is enabled by running:

```
sudo i2c-detect -y 1
```

riveranl, 2017-04-26:
@Gadgetoid i2c is enabled should I download the entire __init__.py file and run the whole program or just take a snippet of it?

Gadgetoid, 2017-11-20:
Did you make any progress with this issue at all?

The `__init__.py` linked was just the one from Explorer HAT which you'll already be running.

closed 2017-11-20 by Gadgetoid