pimoroni/explorer-hat #32: Segmentation fault after running motor script

raspberrypieman, 2017-04-19:
Intermittently, I am getting "Segmentation fault" immediately after running this script:
    import time
    import explorerhat as eh

    eh.motor.one.forwards()
    eh.motor.two.forwards()
    time.sleep(5)
    eh.motor.one.forwards(0)
    eh.motor.two.forwards(0)

(from https://learn.pimoroni.com/tutorial/sandyj/firefly-light)

Here's what it looks like at the command line:
    pi@raspberrypi:~/projects/ExplorerPhat $ python testMotors.py
    Explorer pHAT detected...

    Explorer HAT exiting cleanly, please wait...
    Stopping flashy things...
    Stopping user tasks...
    Cleaning up...
    Goodbye!
    Segmentation fault

What is this trying to tell me?
Should I be worried?

Regards, Martyn

Gadgetoid, 2017-04-24:
This is nothing to worry about, although I should probably get to the bottom of why it happens.

dandrews7396, 2017-07-25:
I may be getting the wrong end of the stick here, but would if you're writing out the same speed to both motors, would you not be better to use:
`eh.motor.forward()
time.sleep(5)
eh.motor.stop()`

patboul, 2017-07-25:
I solved it. Got the same problem, and worse, the pi was freezing after running the test.py script a few time. The default version of Python was the cause. I changed it python 3.4 using the following instructions: https://linuxconfig.org/how-to-change-from-default-to-alternative-python-version-on-debian-linux

closed 2018-02-07 by Gadgetoid