pimoroni/explorer-hat #33: Open Source Hardware

ARKopp, 2017-05-14:
is the hardware opensource as well?

Gadgetoid, 2017-05-17:
Nope. Propeller HAT is our only OSHW board: https://github.com/pimoroni/propeller-hat/tree/master/hardware

closed 2017-05-17 by Gadgetoid

ARKopp, 2017-05-17:
> Nope. Propeller HAT is our only OSHW board: https://github.com/pimoroni/
> propeller-hat/tree/master/hardware

ok thanks. why?

Gadgetoid, 2017-05-17:
It leaves us wide open to a market of third-party clones.

We prefer to double down on excellent software and support, helping people where we can.

ARKopp, 2017-05-18:
when they want to clone you they clone you. This is actually good
marketing. I think your customers will by from you and not the clones.

I can also copy your schematics and puplish the plans. I thought you want
to become the europe adafruit. A big part is that you also

do openhardware and share freely and let people change your designs.