pimoroni/explorer-hat #38: "import explorerhat" gives warning; exits when using pHat

Patronics, 2018-02-24:
When using python 2.7 (possibly python 3 as well, I haven't tested it) in the interactive python console (I haven't tested yet in complete python programs), the line `import explorerhat` triggers the message `Warning, could not find Analog or Touch...
Please check your i2c settings!`, and immediately exits the python console. The pHat does not have touch inputs or analog inputs, which I imagine is the cause of the error.

Patronics, 2018-02-24:
This problem was fixed by installing the latest version though git by following the instructions listed as "development" in readme.md. I suppose that means the problem was already addressed, but the version installed by the pimoroni installer does not yet have this fix.

Gadgetoid, 2018-02-24:
This is curious- it means there's something wrong with your Explorer pHAT because it *should* be finding the analog.

The version as of Git has replaced the badly behaved "exit" with a more specific error, but this wont be triggered until you actually try to use the ADC.

What happens if you:

```python
explorerhat.analog.one.read()
```

And also:

```
sudo i2cdetect -y 1
```

Patronics, 2018-02-24:
Huh. The sudo i2cdetect finds 2c and 40, and the analog read triggers a traceback of 
“File "<stdin>", line 1, in <module>
  File "build/bdist.linux-armv6l/egg/explorerhat/__init__.py", line 667, in read
RuntimeError: Analog is unavailable, check your pHAT/HAT and/or connections!” 
The sparkfun servo hat, and the pimoroni touch phat are also connected to the Pi, if that’s relevant at all.

Gadgetoid, 2018-02-24:
You should be seeing `0x48` on i2cdetect, since that's the address of the ADS1015 on Explorer pHAT.

Touch pHAT is `0x2c` which means your Servo HAT will be the `0x40`.

Explorer pHAT is MIA! Neither of those boards should conflict with it.

How do you have the three boards connected?

Patronics, 2018-02-25:
That’s weird! I have them stacked vertically, with stackable headers. At least some of the explorer phat works, though I’ve only specifically tested the digital output so far.

closed 2018-03-13 by Gadgetoid