pimoroni/explorer-hat #52: Installer does not use the most recent release

AmadeusW, 2022-03-04:
I followed instructions to install explorer-hat on a fresh raspi image (Bullseye version,  kernel 5.10.92-v7+)

At runtime, it failed with `AttributeError: 'AsyncWorker' object has no attribute 'isAlive'` which has been fixed in #50

It looks like the default installation does not use version 0.5.1. explorerhat is already installed (v0.4.2) and therefore it's not updated. Perhaps it would make sense to add a flag to the installer to force an upgrade?

I fixed this with  `pip install --force-reinstall explorerhat==0.5.1` which upgraded `explorerhat-0.5.1` and `cap1xxx-0.1.4`

Gadgetoid, 2022-03-16:
Updated the README to clarify this, and also updated the installer to use `--upgrade` so that it fetches the latest version if there's one already installed.

closed 2022-03-16 by Gadgetoid

RobertJN64, 2022-06-07:
This should also be updated on the PyPI description, it is still pointing users to the apt-get installer which installs an older version.